RenderHeads/UnityPlugin-AVProLiveCamera #1: Alpha channel support

gontz, 2019-05-15:
I use xSplit Vcam and that software takes a camera feed and removes the background (without chroma key). It creates another webcam source. That webcam source has an image format ARGB32. 
Could you implement support for alpha channel?
[image]

AndrewRH, 2019-05-21:
Thanks for the recommendation.  We're adding this feature and it should be released today in version 2.8.0.  We will update the build on our website and on the Asset Store in a few hours.  I can confirm this now works with XSplit-VCam.

Thanks,

closed 2019-05-21 by AndrewRH

stealink, 2021-10-03:
Hi 
I have tried to have the transparency and alpha Chanel in unity single xsplit cam but I can find the setup to do that . 
Do you know how to have the alpha Chanel in the unity view ? 
Thx