RenderHeads/UnityPlugin-AVProLiveCamera #4: Color Accuracy Inconsistent

BlueprintBen, 2019-06-13:
I'm trying to confirm RGB value consistency of camera images across a couple projects and have some questions surrounding that.

To set up a consistent test case, I've employed OBS-VirtualCam and XSplit Broadcaster which each have functionality to pipe out as a DirectShow device, and are configured to display a 0-255 gradient. OBS-VirtualCam produces YUV_422_YUY2 only, and XSplit gives RAW_BGRA32 and YUV_422_YUY2.

[image]

Starting with a Gamma space project, I discovered that the YUV_422_YUY2 feed was displaying significantly different values as you can see from the first image. Thankfully with XSplit I could confirm that the RGB32 option is basically one-to-one. So first question: is this much modification to the source values expected with YUV? I believe there's something about using 16-235 instead of 0-255 that might be acting here?

Moving on to a Linear space project, both images are now displaying worse issues:
[image]

In either format, the gradient is too bright (mid-point is now like 0.75 where it should be .5) and heavy banding is appearing. In YUV format, the black background is being shown as purple for OBS and green for XSplit.

Given these discrepancies, I'd be very interested in hearing whether it's possible to produce consistent RGB output in projects of different color spaces, and whether it's possible to achieve 'loss-less' camera feed access in that case.

Thanks,
Ben

PS: I noticed your color space conversion calculation in the shaders is using pow(rgb, 2.2) while Unity has a far more tuned function called GammaToLinearSpace(rgb) that is worth mentioning.

AndrewRH, 2019-06-13:
Thanks for reporting this! And especially for such a great example.  We'll investigate and report back.

AndrewRH, 2019-06-16:
We have replicated the issue and are looking further...good chance of a fix coming soon.

AndrewRH, 2019-07-03:
Just an update..we're still working on this.  We uncovered a number of issues and kept going down the rabbit hole.  We might break the fixes into pieces so we can at least resolve a few issues faster.  Thanks for your patience.

BenSheftel, 2019-08-08:
Happy to test partial fixes as they're available! The main issue we're having is around chromakeying as you might imagine, but we're looking at workarounds as well. Then need to move onto investigating issues we're having with Decklink cards...

BlueprintBen, 2019-08-08:
Happy to test partial fixes as they're available! The main issue we're having is around chromakeying as you might imagine, but we're looking at workarounds as well. Then need to move onto investigating issues we're having with Decklink cards...

rilez, 2019-08-16:
I've noticed issues with this as well, and am also using it for chroma keying. I'm currently evaluating the demo. As @BlueprintBen mentioned, the issues get worse when using linear color space. I lose a lot of definition in my gradients, particularly blacks. I'll continue to evaluate in the meantime, but any ETA on this?

AndrewRH, 2020-08-18:
We're still looking into this - we just have to find more time :)
Sorry for the delays on this one.

AndrewRH, 2020-08-20:
@rilez  @BlueprintBen  this issue has been resolved in 2.9.0 which was just released.  I am going to close this issue now.  If you have any other issues, or notice any problems with these fixes then please open another issue.

Thank you for reporting this and for your patience while we fixed it.

closed 2020-08-20 by AndrewRH

BlueprintBen, 2020-08-20:
Nice, I appreciate the followup! Can you confirm which color space setting would see changes when updating the plugin? Linear, Gamma, or both?

AndrewRH, 2020-08-20:
@BlueprintBen hmm, I think Gamma space should be the same in both - it was only really linear that we saw the problem in - so we fixed the linear mode.  We also made the YCbCr range selectable which increases options when working with things like OBS (which can have a different range value set).  

Thanks,