RenderHeads/UnityPlugin-AVProLiveCamera #5: FPS decreases on AVPro Live camera plugin

katsuwo, 2019-07-11:
FPS decreases has occurred.
The camera is Logitech C 922.
Right after the start with 1920x1080 MJPG, it is fine.
However, if the camera image changes significantly, the FPS will decrease, and then it will not return.

WIndows10 pro Build17763.615
Unity 2019.1.9f1  
AVPro LiveCam plugin(EVA Version)
Logicool C922 Web-camera　1920 x 1080 MJPG
i7-8700K 32GBmem RADEON Vega56

AndrewRH, 2019-07-17:
Thanks for posting, we'll investigate further...very odd
Does it happen in the other demos too?

AndrewRH, 2019-07-17:
I haven't managed to replicate this issue.  I have a Logitech BRIO webcam and it doesn't slow down like that.  If I completely cover the sensor then it slows down to 15FPS.  This is normal behaviour as the camera is trying to compensate for low light levels by using a longer exposure time (which reduces frame rate).  After under-covering it though, it quickly jumps back to 30FPS.

A few questions / suggestions:
1) Does your FPS rate recover eventually if you stop waving your hand?
2) I notice you're using an old version of the plugin - please upgrade to the latest version (2.8.0)
3) Check if there is a new driver/firmware for your camera - this may fix the issue
4) Try going into the camera settings and disabling things like auto exposure, "right light" etc..see if that helps.

Please let us know how you get on with this.

Thanks,

katsuwo, 2019-07-17:
Dear andrew
Today I conducted the same test as last time.
The result was the same as last time.
Then I changed the camera connection terminal from USB3 to USB2.
As a result, the phenomenon that FPS declines does not occur.
However, even if I set it to 30 FPS, the measured value was 15 FPS.

>1.Does your FPS rate recover eventually if you stop waving your hand?
>2.I notice you're using an old version of the plugin - please upgrade to the latest version (2.8.0)
>3.Check if there is a new driver/firmware for your camera - this may fix the issue

1:no. FPS is not recover until restart application
2;thanks. I will try 2.8.0
3:I can't find new firm and driver for C922

puyingjie, 2019-07-18:
Logitech ce1000 4K，support 60fps，but plugin only display 42fps.
i have a global shutter camera，same to katsuwo's question.

puyingjie, 2019-07-18:
your official website's dowload link is 2.7.5?
Can you provide a 2.8 test version link？

puyingjie, 2019-07-18:
https://v.qq.com/x/page/k09001kkfnv.html

puyingjie, 2019-07-18:
global shutter camera 1920*1080 mjpg 30fps，video test link：
https://v.qq.com/x/page/d09005i5leo.html

Logitech ce1000 1920*1080 mjpg 60fps，video test link：
https://v.qq.com/x/page/k09001kkfnv.html

puyingjie, 2019-07-18:
use UnityPlugin-AVProLiveCamera-Latest-Trial.unitypackage，download from your official website

AndrewRH, 2019-07-19:
Hmmm...thanks for the information.

I think this must be a problem with the camera/driver/firmware..Perhaps ask Logicool/Logitech?

As for:
> Logitech ce1000 4K，support 60fps，but plugin only display 42fps.

To get 60fps you need to make sure the room is brightly lit. Also make sure there aren't any other CPU overheads, so try closing the list of "Modes" in the explorer demo.  Try also making a standalone build.

AndrewRH, 2019-07-19:
> your official website's dowload link is 2.7.5?
> Can you provide a 2.8 test version link？

This link has been fixed now - thanks

puyingjie, 2019-07-19:
win10 provide camer app, qulity is better than 2.8 plugin.

puyingjie, 2019-07-19:
win10 provide camer app, qulity and speed is better than 2.8 plugin. why?
my app is a standalone build.

katsuwo, 2019-07-20:
Yesterday I built OpenCV as a DLL and made it a UNITY plug-in.
The same problem was confirmed there.
The only webcams I have are C922 and Intel RealSense D435,
D435 works with both OpenCV and AVProLiveCam at 1920x1080 @ 30FPS.
In addition, with CameraAPP of Windows 10, both cameras operate normally.

AndrewRH, 2019-07-20:
Interesting...And this was with Skype+Camera app running at the same 1920x1080 resolution?

If so, it might be an issue with using the DirectShow API (which we do, and perhaps OpenCV does too), and it must be an issue with Logitech support.   You can also try to uninstall the Logitech driver - I've heard of some cases where that can help.  Skype and Camera app might be using a new API like Media Foundation.  Sadly I can't replicate the problem here on my Logitech BRIO webcam, so it's difficult to know.

puyingjie, 2019-07-22:
@katsuwo can you send me your dll to me？

puyingjie, 2019-07-22:
@katsuwo can you send me your  OpenCV as a DLL and made it a UNITY plug-in to me？
I want to test my camera c1000e and another camera.
thank you, my email，40595939@qq.com

katsuwo, 2019-07-22:
Hey 
puyingjie

>an you send me your OpenCV as a DLL and made it a UNITY plug-in to me？
NO
I despise you

[image]

AndrewRH, 2019-07-22:
I'm closing this thread as the issue seems related to the webcam/driver.  I hope in the future we can add support for the more modern Media Foundation API to the plugin.  I will create a new issue for this.

closed 2019-07-22 by AndrewRH