RenderHeads/UnityPlugin-AVProLiveCamera #7: Randomly resolution and fps init

Filaind, 2019-08-28:
We have a problem with a camera init.
So, we set a camera index and mode, invoke _liveCamera.Begin();

But, camera starting on randomly resolution and fps.

Example:
Mode 0 = 1920x1080 30fps

But camera start in 640x480 24fps.
If we invoke repeately _liveCamera.Begin(); camera can start normaly in 1920x1080 30fps.

This bug contains on different camera inputs

AVPro Live Camera - Version 2.8.0

AndrewRH, 2019-09-04:
Could you show us the code you're using to set the index and device, and how you're calling Begin()?

Thanks,

Filaind, 2019-09-04:
@AndrewRH 
```
    public void SetDevice(int id)
    {
        cameraInputSettings.cameraIndex = id;
        _liveCamera._deviceSelection = AVProLiveCamera.SelectDeviceBy.Index;
        _liveCamera._desiredDeviceIndex = id;
        _liveCamera.Begin();
        MakeDropdowns();
    }

    public void SetResolution(int id)
    {
        cameraInputSettings.modeIndex = id;
        _liveCamera._modeSelection = AVProLiveCamera.SelectModeBy.Index;
        _liveCamera._desiredModeIndex = id;
        _liveCamera.Begin();
    }
```

AndrewRH, 2019-09-04:
Thanks....I can't see anything wrong with that code!  We're investigating...

AndrewRH, 2020-03-11:
@Filaind  Which webcam device is this happening with?

Chris-RH, 2022-05-13:
Could not reproduce

closed 2022-05-13 by Chris-RH