RenderHeads/UnityPlugin-AVProLiveCamera #8: Decklink video capture is not working on Unity 2018.4.5f1

Avinash9100, 2019-08-30:
Hey i am trying to using Decklink video capture, but its not working , with nikon dslr capture card  blackmagic intensity shuttle
please help me
 - Unity version: 2018.4.5f1
 - AVPro Live Camera version:  Updated version
 - Operating system version:  window 10 64bit
 - Hardware/webcam model: Nikon D5300
[image]

AndrewRH, 2019-09-04:
Hmm

I believe the issue here is that you have to specify the EXACT mode that is used by the device connected to the DeckLink card.

For DeckLink we actually have a dedicated plugin called AVPro DeckLink that will give you a WAY better experience using those devices.  It supports things like automatic input mode detect for example.

In in your example you would need to select one of the 26 modes, but you'll have to figure out exactly which one is required.  It should be the same resolution as your camera is set to output, and then it will be some frame-rate/pixel format variant of that.

If you expand your Game window a bit then you should see a button to expand the list of modes in that demo, then you can try various modes.

Does that help?

AndrewRH, 2020-02-19:
I'm closing this issue as we have no heard back.

closed 2020-02-19 by AndrewRH

Avinash9100, 2020-02-20:
yeha,  close it thanks give me solutions & importance. i like your service.