RenderHeads/UnityPlugin-AVProLiveCamera #12: Build Problem for x86_64 Architecture.

SimfusionTech, 2020-02-08:
**Describe the bug**
Build Problem for x86_64 Architecture.
If x86 is selected for build, the build exe is ok.
If x86_64 is selected, the build output always shows "No Webcam / capture devices found

**Your Setup (please complete the following information):**
 - Unity version: 2019.2.18f1 (64-bit)
 - AVPro Live Camera version: Version2.8.0
 - Operating system version:  windows 10
 - Hardware/webcam model: Logitech C920
 - Video mode (resolution, frame-rate, etc): 1920x1080, 30FPS

In additional to build problem, please also advise below:
(1)	Is there any function to stop camera at script?
I find an "Stop" button at "AV Pro Live Camera.cs" after run and camera enabled.
I want my script able to stop the current selected web camera and then switch to another camera 
after build.

(2)	There is a "Show Config Window" button at "AV Pro Live Camera.cs" in editor.
I want to access the corresponding function in script so that user can press a button to trigger this window after build.

AndrewRH, 2020-02-11:
Hi,

It sounds like you don't have the 64-bit driver installed....but then I presume it works fine in the Unity editor?

For your other questions:
1) yes, you can stop via script:
liveCamera.Device.Stop();
We have plenty of examples in the Demos folder - I recommend having a look through this code.

2) For "Show Config Window" you can use:
liveCamera.Device.ShowConfigWindow();

Cheers,

SimfusionTech, 2020-02-11:
Hi Sir, 

I used to build 64 bit applications for all of my work. 
What specific driver is used for your application?

Thanks,
Re,

⁣取得 BlueMail for Android ​

SimfusionTech, 2020-02-11:
Hi sir,

I mean I can build different scenes under the same project, which make use of webcamtexture.
If any driver is needed for your plugin, please let me know. 
I need to fix it urgently.
Much thanks.

Rex

⁣取得 BlueMail for Android ​

在 2020年2月12日 00:18 於 00:18，"Info@simfusion" <info@simfusion.com.hk>曾寫：

SimfusionTech, 2020-02-12:
Hi Sir,

In addition, i do want to know more how to adopt your plugin. 
There are six scenes in your package and i read all of this.
But i still cant find the function "Device.Stop();". i can only find something like "Close()".
Do you have any doc description your functions? 

Thanks

SimfusionTech, 2020-02-12:
Hi,

I solved the problem. It has some issues with firebase.
The problem could be solved by altering firebase settings.

Thanks

AndrewRH, 2020-02-13:
Yeah I'm glad the problem was solved :)

Our plugin doesn't require any drives, but webcams sometimes require drivers and there can be different drivers for 32-bit and 64-bit apps.

I will close this ticket now.  If you have any other issues please post another ticket.  Thanks,

closed 2020-02-13 by AndrewRH