RenderHeads/UnityPlugin-AVProLiveCamera #14: Support for Network Camera devices in Windows Insider Build 18995 and later

jasonhbartlett, 2020-03-11:
**Is your feature request related to a problem? Please describe.**
Hi Andrew. Recently with Windows 10 Insider Build 18995 it is now possible to add Network Cameras as devices to the OS.  Please see this blog:
[https://blogs.windows.com/windowsdeveloper/2019/10/10/connecting-network-cameras-to-windows-10-devices/](https://blogs.windows.com/windowsdeveloper/2019/10/10/connecting-network-cameras-to-windows-10-devices/ )
Once a network camera is paired, it looks like this in device manager:
[image]
I was hoping that it would be possible for AVProLive to access this camera as it does other web camera devices. (Without too much trouble in updating your plugin :)

I have tried to see if the paired network camera shows up in my Unity app by printing out to the console the GUIDs for every device in AVProLiveCameraPlugin.GetNumDevices(), but unfortunately it doesn't show up with my other cameras.  So this just tells me your plugin doesn't query for these devices as they are somehow different than a typical USB web cam.

**Describe the solution you'd like**
Hopefully it isn't too difficult to update your plugin for this new capability added to Windows?

**Describe alternatives you've considered**
I am considering writing my own plugin for Unity however I imaging that is going to be a large undertaking and it's not something I could do easily.  I have also considered writing a plugin for Unity that would implement all ONVIF profile S functions directly rather than through the Windows 10 OS.  But again, a large and difficult task for me.

If it is possible... I hope you would see the huge potential for this ability to access IP network cameras from a Unity app.  This could be worth $$.

**Additional context**
If you are considering testing this out:
1. You currently need to be running Windows Insider Build 18995 or later.

2. You need to have an IP network camera available that meets ONVIF Profile S specification
[https://www.onvif.org/about/mission/](https://www.onvif.org/about/mission/)
Note: There is a software called DeskCamera that can turn your PC into an ONVIF camera source.  Sources include any monitors or webcams connected to the PC.
[https://www.deskcamera.com/](https://www.deskcamera.com/)
The site lets you request a trial license key.  After installation and activating the license, under Camera Settings change the Camera Mode to "Multiple cameras" and use blanks for the username and password.  Like so:
[image]

3. For cameras that allow you to set a blank username and password, just follow the instructions from the Microsoft blog post and that will pair the network camera with your computer. Basically go to "Bluetooth and other devices" settings, Click the "+" to add Bluetooth or other devices, then select "Everything Else" option.  Any network cameras that can be paired with will show up and you just click it to pair with your computer.

4. For cameras that require setting an ONVIF username and password...For example I have some Axis cameras that let you create an ONVIF username and password like so:
[image]
 I was able to utilize the Windows Universal Samples github repo.  Just clone the repo and build the "Device Enumeration and Paring C# Sample" found here: 
[https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/DeviceEnumerationAndPairing](https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/DeviceEnumerationAndPairing)
And run the "9. Custom Device Pairing" sample like so:
[image]

Anyhow, it would be great if this were possible.  I'll be glad to provide any additional screen captures or logs related to the network camera device on my PC.
If its not possible then I'm bummed.  :(  LOL.  Then I guess I'm looking for any other suggestions for a solution.  Thanks!

jasonhbartlett, 2020-06-29:
Heya! Just poking this to ask if this enhancement is even in the realm of possible. Would still like to see this become a reality to expand on a great product.

AndrewRH, 2020-06-30:
Yes we're very interested in adding this feature.  We're just a bit busy on other projects right now.   Hopefully soon!

jasonhbartlett, 2021-02-05:
Hello again.  It's been six months and I'm back working with this asset again for a work project.  Still hoping this enhancement is in the queue to be looked at sometime.  Cheers!

kahnivore, 2021-02-09:
Hi @jasonhbartlett 
Thanks for the reminder - this is still something we'd really like to do! We're a small team and our focus over the last bit has been the AVProVideo2 updates, but hoping to get to this in the upcoming months. Will keep you posted :)