RenderHeads/UnityPlugin-AVProLiveCamera #15: Reported Name/GUID for Software Device incorrect when Decklink hardware present

BlueprintBen, 2020-03-20:
The issue I'm seeing is around the ability for AVProLiveCamera to report a GUID for a device, specifically some software devices. Rather than an `@device:sw:` style GUID, the plugin reports one of the Decklink profile names, and in the other, the device name was a Decklink 

 - Unity version: 2019.2.21
 - AVPro Live Camera version: 2.8.0
 - Operating system version: Windows 10
 - Hardware/webcam model: OBS-VirtualCam (OBS plugin), XSplit VCam, "Decklink Video Capture"

**To Reproduce**
1. Install a Black Magic Decklink card in your machine.
2. Install [XSplit VCam](https://www.xsplit.com/vcam) or [OBS + OBS Virtual Cam](https://obsproject.com/forum/resources/obs-virtualcam.539/) and run it.
3. Run CameraExplorerDemo

Note the enumerated device information in the Unity Console Log.

[image]

[image]

[image]

[image]

AndrewRH, 2020-03-22:
Hmm interesting.....I'm not sure if this is a bug or not but we'll investigate...

BlueprintBen, 2020-03-22:
Yeah was a weird one. Maybe it's mis-handling a null response for GUID and ends up reusing the last ID or causing the indexes for GUIDs be off or something. I'd obviously much rather if we don't know a GUID that it just returned empty or the name instead of introducing a weird interaction with other HW. 
Thanks!

AndrewRH, 2020-03-22:
Hmm that's strange..
Which version of OBS etc are you using?
I'm using OBS Studio 25.0.1 and OBS Virtual Cam 2.0.4 and XSplit Vcam 1.0.2003.301

I tested with those and the GUID enumeration worked for me....

BlueprintBen, 2020-03-22:
I had just updated Vcam to latest. OBS Virtual Cam probably a bit older. But from memory this was not an issue until I plugged in my Decklink 8K Pro card (even when I had a Decklink 4K Mini installed). I've been testing the AVProDecklink plugin in conjuction; I should have mentioned that the plugin is actually imported in the project I've been testing with, but not activated with a DecklinkManager in the scene. I suppose that could factor in somehow.

I should have time this week to try out the same test in other HW configurations (no Decklink, other Decklink) and in a clean project rather than my working one. My apologies for firing this up before having collected that data as well!

closed 2022-05-13 by Chris-RH