RenderHeads/UnityPlugin-AVProLiveCamera #16: Device Settings aren't correct if device isn't running

BlueprintBen, 2020-05-04:
This bug concerns the access of the DirectShow setting values associated with a connected device that is **not** 'running'/active/playing. I have a desire to launch the DirectShow "config window" associated with a camera and then read back/save the user's parameter values (Brightness, Saturation, Focus, Enable Auto Focus, Exposure, Enable Auto Exposure, etc) without actually activating the image stream and therefore allowing a different process to access it. 

When the application responsible for this process has the camera playing, `AVProLiveCameraSettingFloat.Update()` correctly populates `_currentValue` and `_isAutomatic`. When the camera isn't playing, Update() doesn't provide new values after they've been changed via the config window/externally.

**My Setup:**
 - Unity version: Unity 2019.3.11f1
 - AVPro Live Camera version: 2.8.0
 - Operating system version: Windows 10 
 - Hardware/webcam model: Logitech C920

**To Reproduce**
1. Run Camera Explorer Demo
2. Toggle on Update Settings for a test camera
3. Launch OBS and add the same test camera as a video input
4. In OBS, open the test camera's properties and from there select Configure Video.
5. Modify any of the camera properties.
6. Note that the Camera Explorer Demo UI doesn't reflect the changes in values, either live or after hitting OK on the config window. If the camera is running (eg: hit Start after step 1), the UI controls for Gain/Exposure/etc will update live as the Config Window sliders are manipulated.

I realize this is my third open issue but I thought I'd raise whatever flag I could and have them get processed naturally. This feels like it might be an internal assumption that could be easily removed for some significant benefit (at least for my use case).

AndrewRH, 2020-05-07:
Thanks we appreciate the feedback.  I will see about reproducing this and your other issues and get back to you.
Thanks,

AndrewRH, 2020-06-02:
I have reproduced this.  Investigating a fix...