RenderHeads/UnityPlugin-AVProLiveCamera #17: Issues with UGUI component

AzeExMachina, 2020-05-27:
**Describe the bug**
I'm having issues when testing UGUI component, specifically when changing source.

**Your Setup (please complete the following information):**
 - Unity version: 2018.4.22f LTS
 - AVPro Live Camera version: 2.8.0
 - Operating system version: Windows 10 Pro
 - Hardware/webcam model: Matrox Capture Card
 - Video mode (resolution, frame-rate, etc): 1440p, 144hz but I'm having issues with every monitor

**To Reproduce**
Steps to reproduce the behavior:
1. Open GUI Demo
2. Click on any source, the source gets rendered
3. Change source to another one
4. The ugui component stops rendering and doesn't apply the new texture, what is seen is an old frame. To work around this I have either to manually change the resolution everytime I change source or to disable/enable the UGUI component.

AndrewRH, 2020-05-29:
Thanks for reporting this issue.  We're looking into it and will get back to you soon.

Thanks,

AndrewRH, 2020-05-29:
Hmm, we're not able to replicate the issue you describe.  When running the GUIDemo scene and switching devices our GUI component is switching immediately.

I'm not sure why this is...

We're testing using the same Unity version as you.  I wonder if there could be something else that's different in your setup.  Are you using any packages or SRP etc?

AzeExMachina, 2020-05-29:
I have no render pipeline, no preview packages also. I'll try the package with a fresh project with just this plugin and I'll let you know

AndrewRH, 2020-06-03:
Hi @AzeExMachina  did you manage to test it, or is the issue resolved?

Thanks,

AzeExMachina, 2020-06-04:
Hi @AndrewRH I was able to test this on a new fresh project, I'm getting the same issue, it's not always happening but in a minute of testing I was able to replicate it a couple of times. I have imported everything inside the plugin it's on your site, it's the demo, don't know if it's different from the actual released plugin aside from the watermark. Can it be an issue importing both the x86 and x64 dlls?

AndrewRH, 2020-06-04:
> Hi @AndrewRH I was able to test this on a new fresh project, I'm getting the same issue, it's not always happening but in a minute of testing I was able to replicate it a couple of times. I have [...]

Thanks for testing again.  Yes the demo version just has a watermark as the only difference, so I can't imagine that affecting anything.   

Hmm I'm not sure what you mean by importing both DLLs.   It should only import either 64-bit or 32-bit DLL, depending on your your app (or Unity) is built with.  

I wonder if it's something specific related to your Matrox / OBS driver / versions?   Are they up to date?  Are you able to replicate this when using simple webcams?

I tested here with my webcams and OBS but still haven't managed to replicate it.

After checking the above, perhaps the next step is for you to send us your minimal test case project (unitysupport@renderheads.com)?

Thanks,

AzeExMachina, 2020-06-05:
> I wonder if it's something specific related to your Matrox / OBS driver / versions? Are they up to date? Are you able to replicate this when using simple webcams?

The driver are all update, obs is running with the latest version and the matrox drivers too. Also, I had the same issues with a usb Webcam.

Thanks for your effort, I'll try sending you the minimal test case project to the your support, citing this issue.

AndrewRH, 2020-06-08:
Thanks for sending over the test case.    I think we've replicated the issue.  I've attached here a new script for you to try - please copy it over the old one (in /AVProLiveCamera/Scripts/Components)

Let me know if that solves the problem for you.  And thanks for your patience!

[AVProLiveCameraUGUIComponent.zip](https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/files/4745861/AVProLiveCameraUGUIComponent.zip)

AzeExMachina, 2020-06-09:
Thanks @AndrewRH This is now working perfectly! Thank you very much. Was it simply an issue that I could solve by setting the material to dirty? I thought I tried that but it was not working from outside the UGUI script, anyway I'll keep your script. Awesome support!

AndrewRH, 2020-06-09:
Great I'm glad it's solved :) It was a genuine bug.  Yes still don't 100% understand why, but when the texture changes we need to set the material to dirty...  We will include this fix in the next release of the plugin.  I will close this issue now, but thanks for reporting the issue and for your patience while we investigated.

Cheers,