RenderHeads/UnityPlugin-AVProLiveCamera #19: FLIR Video USB Webcam crashes Editor

XPav, 2020-06-03:
**Describe the bug**
The Unity Editor crashes when using AVProLiveCamera to read a FLIR Video (USB) camera.  The camera functions in AMCap, Windows Camera, etc. 

WebcamTexture also crashes, with the very similar stack trace.

**Your Setup (please complete the following information):**
 - Unity version: 2019.3.15f1
 - AVPro Live Camera version:  Version 2.8.0
 - Operating system version:  Windows 10
 - Hardware/webcam model: FLIR BOSON 640
 - Video mode (resolution, frame-rate, etc): 640x480, 30/60fps.  Only outputs I420, Y16, NV12 NV21

**To Reproduce**
Steps to reproduce the behavior:
Use the demo camera scene, open the camera.

**Logs**
Key portion of the stack trace:
```
========== OUTPUTTING STACK TRACE ==================

0x000000007F8411A9 (lvcod64) DriverProc
0x00007FFFD280292C (MSVFW32) ICSendMessage
0x00007FFFD2802AC4 (MSVFW32) ICOpen
0x00007FFFD2803368 (MSVFW32) ICLocate
0x00007FFFC2557F06 (quartz) AMGetErrorTextW
0x00007FFFC2612B34 (quartz) AMGetErrorTextW
0x00007FFFC2527326 (quartz) AMGetErrorTextW
0x00007FFFC2559045 (quartz) AMGetErrorTextW
0x00007FFFDBE63564 (qcap) DllCanUnloadNow
0x00007FFFDBE63722 (qcap) DllCanUnloadNow
0x00007FFFDBE6385B (qcap) DllCanUnloadNow
0x00007FFFDBE63482 (qcap) DllCanUnloadNow
0x00007FFFC252FFE3 (quartz) AMGetErrorTextW
0x00007FFFC253F3DC (quartz) AMGetErrorTextW
0x00007FFFC253F9D4 (quartz) AMGetErrorTextW
0x00007FFFC253FBC7 (quartz) AMGetErrorTextW
0x00007FFFC253F803 (quartz) AMGetErrorTextW
0x00007FFFC253F695 (quartz) AMGetErrorTextW
0x00007FFFC253F301 (quartz) AMGetErrorTextW
0x00007FFFDBE70E92 (qcap) DllCanUnloadNow
0x00000227C3E99E45 (AVProLiveCamera) UnityRenderEvent
  ERROR: SymGetSymFromAddr64, GetLastError: 'Attempt to access invalid address.' (Address: 00000227C3E73908)
0x00000227C3E73908 (AVProLiveCamera) (function-name not available)
0x00000227C7A9A3AF (Mono JIT Code) (wrapper managed-to-native) RenderHeads.Media.AVProLiveCamera.AVProLiveCameraPlugin:StartDevice (int,int,int)
0x00000227C7A99333 (Mono JIT Code) [C:\dev\saipanproposal\Assets\AVProLiveCamera\Scripts\Wrapper\AVProLiveCameraDevice.cs:315] RenderHeads.Media.AVProLiveCamera.AVProLiveCameraDevice:Start (RenderHeads.Media.AVProLiveCamera.AVProLiveCameraDeviceMode,int) 
0x00000227C7A981EB (Mono JIT Code) [C:\dev\saipanproposal\Assets\AVProLiveCamera\Scripts\Components\AVProLiveCamera.cs:145] RenderHeads.Media.AVProLiveCamera.AVProLiveCamera:Begin () 
0x00000227C568831B (Mono JIT Code) [C:\dev\saipanproposal\Assets\AVProLiveCamera\Demos\Scripts\QuickDeviceMenu.cs:75] RenderHeads.Media.AVProLiveCamera.Demos.QuickDeviceMenu:OnGUI () 
0x000002284452CFE8 (Mono JIT Code) (wrapper runtime-invoke) object:runtime_invoke_void__this__ (object,intptr,intptr,intptr)
0x00007FFFC3CDCBA0 (mono-2.0-bdwgc) [c:\build\output\unity-technologies\mono\mono\mini\mini-runtime.c:2809] mono_jit_runtime_invoke 
0x00007FFFC3C62112 (mono-2.0-bdwgc) [c:\build\output\unity-technologies\mono\mono\metadata\object.c:2921] do_runtime_invoke 
0x00007FFFC3C6B10F (mono-2.0-bdwgc) [c:\build\output\unity-technologies\mono\mono\metadata\object.c:2968] mono_runtime_invoke 
0x00007FF6651A5BDE (Unity) scripting_method_invoke
0x00007FF66519F94D (Unity) ScriptingInvocation::Invoke
0x00007FF662A348B9 (Unity) MonoBehaviourDoGUI
0x00007FF662A36DA0 (Unity) IMGUIModule::MonoBehaviourDoGUI
0x00007FF66516AAB4 (Unity) MonoBehaviour::DoGUI
0x00007FF662A2EA1D (Unity) GUIManager::DoGUIEvent
0x00007FF662A35A32 (Unity) GUIManager::SendQueuedEvents
0x00007FF662A3158A (Unity) `GUIManager::InitGUIManager'::`2'::PreUpdateIMGUISendQueuedEventsRegistrator::Forward
0x00007FF664C32A08 (Unity) ExecutePlayerLoop
0x00007FF664C32ADD (Unity) ExecutePlayerLoop
0x00007FF664C37D1C (Unity) PlayerLoop
0x00007FF6630054AC (Unity) PlayerLoopController::UpdateScene
0x00007FF663003438 (Unity) Application::TickTimer
0x00007FF663966E20 (Unity) MainMessageLoop
0x00007FF663970F18 (Unity) WinMain
0x00007FF666942692 (Unity) __scrt_common_main_seh
0x00007FF845277BD4 (KERNEL32) BaseThreadInitThunk
0x00007FF84544CE51 (ntdll) RtlUserThreadStart

========== END OF STACKTRACE ===========

```

XPav, 2020-06-03:
I can "fix" this problem by deleting lvcod64.dll out of Windows/System32.

AndrewRH, 2020-06-04:
Hi, thanks for reporting the issue!

That's a very interesting fix....deleting the Logitech file...  Actually I remember seeing a similar issue before, and I wrote some code to prevent any such enumeration for one of our other plugins:

[image]

We could perhaps look at adding this exclusion to the AVPro Live Camera plugin too...

Do you know what the file version is of your file?  I'm wondering whether I can just exclude old files as I don't seem to be having trouble with mine:

[image]

Thanks,

XPav, 2020-06-04:
That's the same version that I deleted.   

This doesn't affect the "normal" Webcam that I'm using. The camera is a FLIR Boson Infrared camera that doesn't output RGB-format frames, which is probably exposing all sorta of weird code paths.

AndrewRH, 2020-06-04:
Ahh good point.  Yes it could be trying to use the I420 mode and loading the Logitech filter to help it do that...

Could you please try in the Camera Explorer demo scene to untick this option:
[image]

That should allow our own I420 converter to be used instead of replying on internal DirectShow filters to handle conversion.

Does that make any difference?

Thanks,

XPav, 2020-06-04:
If I check that box I get "No webcam/capture devices found" (and I've got the FLIR camera, a standard logitech camera, and a decklink capture device normally).

I've got my workaround for now, as goofy as it it.  Thanks for the help!

AndrewRH, 2022-05-13:
Need to decide whether to apply this fix.  I think we should and expose an option perhaps to allow this DLL to be enumerated.  Though I think it's only being used for colour-conversion so not enumerating it may not prevent this crash