RenderHeads/UnityPlugin-AVProLiveCamera #20: AVPro LIVE capture FPS low on Elgato

Digitrevx, 2020-07-02:
Once installing the plugin, using a elgato HD60S capture card into Live PRO FPS in the inspector shows below 25fps. Visual inspection indicates the fps is quite bad. Not even 30fps.

The configuration for the video feed within Live Pro shows the elgato is running at 60fps. Elgato's game capture software indicates it is also running at 60fps. If live pro's own component is indicating low fps it seems this issue is contained in Live Pro at the moment.

**Your Setup (please complete the following information):**
 - Unity version: 2019.3.14ft
 - AVPro Live Camera version: 2.8.2 -June15th 2020
 - Operating system version:  Windows 10
 - Hardware/webcam model: Elgato HD60 S
 - Video mode (resolution, frame-rate, etc): 1920x1080 @ 60hz

**To Reproduce**
Steps to reproduce the behavior:
1. Open Unity 2019.3.14.
2. Download the AV Pro Live plugin and add to assets.
3. Create all necessary components to receive a live feed set AVPro Live to replace the main texture of a custom material / shader on a 3d object. (Standard shader is fine)
4. Connect a Elgato HD60 S to machine and install any drivers nessesary. (Recomend downloading their "Game Capture for Windows" software since it comes with all driver)
5. Hit play in Unity and observe the performance of the video or check the camera feed component info for a live fps value of AVPro Live.

I can update with video or screen shots at a later point.

AndrewRH, 2020-07-03:
Hi @Digitrevx 

Thanks for the report.  It sounds like we may have to get hold of an Elgato to test with.

In the meantime, could you please test for us using [GraphStudioNext](GraphStudioNext) (there are links to the EXE downloads in the readme).  You can use this software to add the Elgato to the graph and then render it.  I'd be interested to know whether this works any better.

Thanks,

AndrewRH, 2020-07-06:
Hi @Digitrevx  if you could answer the questions above it will really help us to solve this issue for you.

Thanks,

Digitrevx, 2020-07-06:
Hi Andrew,
The hyperlink didn't work but I ended up googling and got this.
https://www.videohelp.com/software/GraphStudio
I'm not sure if this is the same thing.

Anyways the elgatos are at my studio using company equipment. So I will try
to get down there and see if I can run this software to test with. I don't
particularly like installing random software on company machines however.

Cheers,
Digi

Digitrevx, 2020-07-07:
Hi Andrew,

I opened GraphStudioNext 64 on one of the machines. I have absolutely no
idea what I'm doing from there to give any kinda info about fps.

Cheers,
Digitrevx

AndrewRH, 2020-07-29:
@Digitrevx 

This issue should be resolved in the 2.8.4 release coming out today.  If you have a chance please could you test it and let us know whether this solves the issue for you.

Thanks,

AndrewRH, 2020-07-29:
The trial version can be found here:
https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/releases

The full version will be updated on the Asset Store later today.

Thanks,

Digitrevx, 2020-07-30:
Thank you! I will test asap!

Digitrevx, 2020-08-06:
Hi Andrew,

I don't have the best news. As soon as I updated the image off the elgato
displays black. I've tried exiting the project deleting it and the meta
file in the asset root and doing a fresh install off the unity asset store.
I've attempted to unplug the HD60S many times and restarting unity after
installing the LivePro cam package. Secondly the display fps seems worse
but I guess that is second priority under getting any image at all.

Also does this support HDPR?

I kinda need this to be working again asap. We use the plugin a lot for our
projects.

Cheers,
Trevor Taylor

Digitrevx, 2020-08-06:
Hi Andrew,

I just wanted to send a update since it was meaningful. I changed clock
mode to default and back to none and it started working. I know it doesn't
make sense. However now that it is working. I'm noticing a greater loss of
fps then before.
It was showing 22-25fps now struggling to hit 17fps. I'm not sure what is
happening but it seems to be getting 60fps if I use another tool like obs
or elgatos own software to observe the fps. I apologize this is a pain!

Attached are screen grabs from the project.
[image: Unity_sTBAdKtbTG.png]
[image: Unity_LmoaLw1sIE.png]

AndrewRH, 2020-08-06:
Thanks for the update!  Could you send the images on GitHub - they aren't coming through via email...
https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/issues/20

Thanks,

Digitrevx, 2020-08-09:
Hi Andrew I'll attach them here!  Sorry about the wait! Everything is a juggle ><

[image]
[image]

AndrewRH, 2020-08-11:
Hi @Digitrevx 

Thanks for the screenshots.  I'm not sure why it's not able to achieve 60fps.  We will order the Elgato HD60 S, do some testing and get back to you.

Thanks,

Digitrevx, 2020-08-11:
I'm very much looking forward to it. I'll try to see if I can experiment on
my end as well.

AndrewRH, 2020-08-13:
The Elgato arrives tomorrow so we'll start looking into this soon.

Thanks,

AndrewRH, 2020-08-14:
Hmm...I've connected the Elgato HD60S and I have the 1920x1080@60 signal and it's working perfectly for me.   I get a good FPS....In the demo scenes I do have press spacebar to hide the debug mode selection UI as this eats up a fair amount of frame rate.  And it can also help to not have the LiveCamera element selected in the inspector..

I'm not sure how I can replicate the issue you're having with the very low FPS.  Any ideas?

Perhaps you could share your machine spec and also some screenshots of the unity profiler to see if there are any clues there about where the time is going.

Thanks,

Digitrevx, 2020-08-14:
I'll attempt to look at the profiler soon. This unity scene and set up is
not at home and easily accessible.

The machine is dedicated to just unity and specs is: 3950x 2080ti, 16 gigs
of ram 3200mhz, gen4 pcie m.2 1tb, asus rog crosshair viii formula
motherboard, asus 850w seasonic psu.

Overall I doubt the machine specs are the issue at the moment.

The elgato is connected to a proper usb 3.1 connector on the back of the
machines IO.

Digitrevx, 2020-08-16:
I've been messing with the scene some and for sure noticing a 5-8fps jump
when the preview is shut off like you mentioned. However the fps seems
directly linked to the availability of a cpu's main thread. Things like
camera's rendering, real time reflection probe and real time GI and
the post processing stack can easily take the plugin from 60fps to
20-30fps. Running the plugin, in a empty scene does for sure yield about
60fps but not so much in a scene doing other things. Our scene using a
3950x at 4.2ghz running about 5.5ms on the main thread rendering will be at
best 32fps using the plugins own counter.

Doing other things like streaming in bvh data to a character adds a bit
more. The cpu will reach 6.5ms and the fps drops to 25-27fps.

I've removed post processing from both the camera used for AVpro as well as
the main camera and removed every layer from their culling. It has helped
some to get it up to 25-27. Up from 19fps. But I'm looking for a solid 30+.

AndrewRH, 2020-08-17:
> I've been messing with the scene some and for sure noticing a 5-8fps jump when the preview is shut off like you mentioned. However the fps seems directly linked to the availability of a cpu's main [...]

Thanks for testing that... I'll have another look at the code - but AVPro Live Camera is already pretty well optimised.  We don't do much on the CPU side except upload the frames from system memory to GPU texture.   After that we use a pixel shader to convert from YUV format to a RGB texture.

If you disable the AVPro Live Camera feed in your scene and disable vsync - how fast does it render?  I'm just trying to work out how much headroom we have left for the camera processing after your scene has rendered.  If you're getting not much over 60 fps, then running the camera feed could bring it to say 53 and since that would cause a vsync loss it will cause scene rendering to run at 30, which would then cause the frame rate of the camera feed to be lower too (25 say) since it's updated less frequently.

As I said I will look again at the code to see if there is somewhere to improve performance and let you know.

Thank,

AndrewRH, 2020-08-19:
@Digitrevx 
I'm not sure if this will fix your frame rate issue but we're working on some changes.  Currently the plugin uses the Preview pin of the Elagto - we're changing things to use the Capture pin by default now and adding an option to prefer the Preview pin for backward compatibility.  Again I don't know if this will make a difference to you, but it's something we're fixing for another reason and there's a chance it will help you.  I will let you know when the new version is out.

Thanks,

AndrewRH, 2020-08-20:
@Digitrevx 2.9.0 has just been released - could you see whether it solves the low frame rate issue?

Thanks,

closed 2021-10-04 by kahnivore