RenderHeads/UnityPlugin-AVProLiveCamera #22: Missing DLL free version

Xela124, 2020-11-16:
**Describe the bug**
Hello,
I want to try your plugin before buy it. I downloaded the free trial version on your website and import it into Unity.
But Unity says me that He missing DLL File. 

Where can I find this DLL since I only have the package you give ? 

Thank you

**Your Setup (please complete the following information):**
 - Unity version: Tested with 2019.x & 2020.x
 - AVPro Live Camera version: Latest on your web site
 - Operating system version: Windows 10 - 64 bits
 - Hardware/webcam model: Logiech HD Pro C 920

**Screenshots**
[image]

Xela124, 2020-11-16:
Sorry for the quick up but I notice something : 

Here are the file (free version) when I emport to Unity : 
[image]

Here are the file (On asset store "Package content") : 
There are the Dlll files that are missing in the free version
[image]

AndrewRH, 2020-11-16:
Hi,
I just downloaded the trial version and it seems fine to me:
[image]
Where are you downloading it from?  The link should be:
http://downloads.renderheads.com/2020/UnityPlugin-AVProLiveCamera-Latest-Trial.unitypackage

Thanks,

Xela124, 2020-11-17:
Hi Andrew,
Thank for your answer. 
I downloaded it from your site. I try your link and now I can import the plugin folder :) 
Thnak for your help

AndrewRH, 2020-11-17:
Great, I'm glad that's solved.  I'll close this issue now.

closed 2020-11-17 by AndrewRH