RenderHeads/UnityPlugin-AVProLiveCamera #23: cause Unity crash when use YUV_I420

Araraki, 2020-11-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Your Setup (please complete the following information):**
 - Unity version: 2019.4.14f1c1
 - AVPro Live Camera version: Free Trial Version 2.9.0
 - Operating system version: Windows 10 1909
 - Hardware/webcam model: Logitech HD Webcam C310
 - Video mode (resolution, frame-rate, etc): 1280x960@30hz

**To Reproduce**
Steps to reproduce the behavior:
1. Import AVProLiveCamera(Free Trial Version)
2. Play demo scene "CameraExplorerDemo"
3. Click "Modes -> [any YUV_PLANAR_I420 mode]"
4. Crash

**Logs**
I checked the Unity editor error log but I don’t know what caused it, I pasted it below:

`[AVProLiveCamera] version 2.9.0t initialised. NVIDIA GeForce RTX 3070`
`UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)`
`UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])`
`UnityEngine.Logger:Log(LogType, Object)`
`UnityEngine.Debug:Log(Object)`
`RenderHeads.Media.AVProLiveCamera.AVProLiveCameraManager:Init() (at Assets\AVProLiveCamera\Scripts\Components\AVProLiveCameraManager.cs:89)`
`RenderHeads.Media.AVProLiveCamera.AVProLiveCameraManager:Start() (at Assets\AVProLiveCamera\Scripts\Components\AVProLiveCameraManager.cs:74)`
`(Filename: Assets/AVProLiveCamera/Scripts/Components/AVProLiveCameraManager.cs Line: 89)`
`Stacktrace:`
`  at <unknown> <0xffffffff>`
`  at (wrapper managed-to-native) RenderHeads.Media.AVProLiveCamera.AVProLiveCameraPlugin.StartDevice (int,int,int,int,bool) [0x00018] in <5353ec25d4f547fa9850b1996f7997c7>:0`
`  at RenderHeads.Media.AVProLiveCamera.AVProLiveCameraDevice.Start (RenderHeads.Media.AVProLiveCamera.AVProLiveCameraDeviceMode,int) [0x00027] in C:\uname\tan\Desktop\labs\nvBroadcast Test\Assets\AVProLiveCamera\Scripts\Wrapper\AVProLiveCameraDevice.cs:354`
`  at RenderHeads.Media.AVProLiveCamera.AVProLiveCamera.Begin () [0x000ba] in C:\Users\uname\Desktop\labs\nvBroadcast Test\Assets\AVProLiveCamera\Scripts\Components\AVProLiveCamera.cs:180`
`  at RenderHeads.Media.AVProLiveCamera.Demos.QuickDeviceMenu.OnGUI () [0x00159] in C:\Users\uname\Desktop\labs\nvBroadcast Test\Assets\AVProLiveCamera\Demos\Scripts\QuickDeviceMenu.cs:75`
`  at (wrapper runtime-invoke) object.runtime_invoke_void__this__ (object,intptr,intptr,intptr) [0x00020] in <9577ac7a62ef43179789031239ba8798>:0`
`=================================================================`
`Got a SIGSEGV while executing native code. This usually indicates`
`a fatal error in the mono runtime or one of the native libraries `
`used by your application.`
`=================================================================`
`Crash!!!`

**Screenshots**
[Screenshots](https://i.ibb.co/T4PJyNL/ss.png)

AndrewRH, 2020-11-17:
Thanks for reporting this issue, we will investigate and get back to you.

phields, 2021-01-29:
> Thanks for reporting this issue, we will investigate and get back to you.

I think it's the same problem with:
https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/issues/28

AndrewRH, 2021-01-29:
@phields thanks for reporting this.  We will look into it soon.

AndrewRH, 2021-02-02:
We have reproduced this issue.  It seems to be related to buggy Logitech drivers.

See this thread here where they discuss it:
https://forum.doom9.org/archive/index.php/t-177049.html

It seems that Logitech overrides the Microsoft colour conversion routines for I420 (and NV12 perhaps too), but it has a buggy implementation that crashes.

We are working on a solution for this where we implement the conversion ourselves via a shader.

@Araraki  does that Logitech HD Webcam C310 have a NV12 mode, or how are you getting that one?  I don't have a device here that is exposing that mode for me to test...

Hopefully we can release a fix for this soon.

Thanks,

AndrewRH, 2021-02-02:
@Araraki  sorry i didn't see your screenshot until now - thanks that's very useful!

I see the NV12 is coming from OBS so we will test that.

Thanks,

AndrewRH, 2021-02-17:
Hi @phields @Araraki,

This issue has been fixed now in the latest release (2.9.1).  Please do let us know if it resolves the issue for you.

We had to add native support for I420 pixel format, because a Logitech DLL was buggy and caused the format conversion to crash.  

We also noticed that some versions of the OBS Virtual Camera are quite buggy.  Selecting NV12 pixel format caused not frames to be produced, and switching between formats caused an internal stack overflow crash.  Also starting the virtual device without selecting a mode didn't work - a mode must be selected first.

We found another virtual webcam for OBS that appears in the drop-down tools menu and that created cameras called "OBS Camera".  This plugin seems much more stable, so we recommend using this.

I will close this issue now but if you are still having issues please create a new issue.

Thanks,

closed 2021-02-17 by AndrewRH