RenderHeads/UnityPlugin-AVProLiveCamera #25: Camerafeed seems to freeze after a while

Kuinshi, 2020-11-23:
Hello!
Thanks for you great plug-in! Thus far it worked awesome for us, sadly we discovered a bug lately and are unsure of where exactly it stems from... 
We use Logitech Brio and have it running for the full duration of the application, but occasionally it seems like the camera feed is crashing after a while now. The camera's light is still truned on as if it is still running but the Camerafeed is empty. Unfortunately the logs do not state an error. So we were wondering if you have any ideas that could cause such an issue.
Furthermore, while we have the camera running permanently, we are not using the Feed permanently, would it be better to turn the camera on and off everytime, disregarding the gain in performance - as performance is not an isse, we thought that would be more error-prone, that's why we left it running.

Unity Version: 2019.3.2f1
AVPro Live Camera version: 2.8 - According to the Manual
Operating System: Windows 10
Webcam: Logitech Brio
Run stats: 3840x2160 Rest, Frame Rate set to 0 so highest will be used

Would be very thankful for any tips and hints!

Kind regards
Ken

AndrewRH, 2020-11-24:
If it's causing the application to crash, then it's probably not an issue we can resolve as it's likely a driver issue.  You may want to try upgrading the driver.

If it's just the feed that freezes and stops updating the texture, you should be able to detect this by monitoring the texture update index (or checking the display rate FPS value).  If this happens you could detect this and then try restart the device, or restart the app.  We have had a similar issue in the past where the Brio overheated.  It needs to be well ventilated as it gets quite hot.

If your app runs 24/7 then it may also be a good idea to schedule a 3am reboot as well as in the real world this 
is generally safer than just leaving it running (especially in Windows).

As for turning the camera on/off as it's needed - I'm not sure, it often depends on how the driver was implemented so you would have to test this.  If the camera is overheating or running into some internal issue then turning it off is probably a good idea.  

I hope this helps?

farazkhalidsiddiqui, 2021-01-06:
I am experiencing same issue, camera feed freezes after some time randomly in few cameras.

Chris-RH, 2022-05-13:
Is anyone still having an issue with this or have you found a workaround?

Kuinshi, 2022-05-13:
> 

We worked around it by deactivating the camera feed and reactivating it only when necessary, instead of running it persistently. Works for this application, but is obviously no solution for applications that rely on a persistent camera feed!

Chris-RH, 2022-05-16:
Hi, @Kuinshi. Thank you for your reply. Approximately how long does it usually take before the feed stops working for you? Does it vary much?

Chris-RH, 2022-05-20:
Closing as we could not reproduce this in testing. Please reopen if this is still an issue that is occurring.

closed 2022-05-20 by Chris-RH

gekkyo, 2024-12-04:
Hi, I know this issue is closed, but I am experiencing the same problem and am working on detecting when the freeze occurs in order to automatically reset the device.

In that case, what would be the best approach to detect the freeze?
As Andrew mentioned, I plan to monitor the texture update index. To do this, is checking AVProLiveCamera.outputTexture.updateCount correct?

Chris-RH, 2024-12-05:
Hi @gekkyo 
It might be helpful if you open up a new issue and fill in all the details so that it is relevant to your use case.