RenderHeads/UnityPlugin-AVProLiveCamera #28: logitech c270 crush every time

phields, 2021-01-29:
**Describe the bug**
A clear and concise description of what the bug is.

**Your Setup (please complete the following information):**
 - Unity version: 2019.3.5
 - AVPro Live Camera version: 2.9.0
 - Operating system version: win10
 - Hardware/webcam model: logitech c270
 - Video mode (resolution, frame-rate, etc): null

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'

2. Click on '....'
3. Scroll down to '....'
4. See error

**Logs**
If applicable, add error logs to help explain your problem.
[error.log](https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/files/5894218/error.log)
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Videos**
If applicable, add a copy of your video or the URL

phields, 2021-01-29:
same issues on logitech c920

AndrewRH, 2021-02-02:
I am closing this as it's a duplicate of #23 - we will follow up in there

closed 2021-02-02 by AndrewRH