RenderHeads/UnityPlugin-AVProLiveCamera #30: YUV_I420 not work with AVProLiveCameraPlugin.GetFrameAsColor32

phields, 2021-03-02:
https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/issues/23
after this update,
I can't use logic C270 camera with AVProLiveCameraGrabber

kahnivore, 2021-03-04:
Hey @phields 

What version of Unity/OS are you using?

phields, 2021-03-09:
> Hey @phields
> 
> What version of Unity/OS are you using?

CPU	AMD Ryzen 7 1700X Eight-Core Processor            3.40 GHz
RAM	32.0 GB
Windows 10 20H2
Unity version 2020.1f

phields, 2021-03-09:
I tested this on different PCs, CameraExplorerDemo works fine.
but there is no camera readback in CameraGrabDemo :(

kaushal135, 2021-03-26:
I am also having this issue with unity 2019.2.9.

phields, 2021-03-31:
I believe this is a bug, the old version does not support YUV_I420, but in other modes (RGB) can be displayed normally @kahnivore

AndrewRH, 2021-04-01:
We will implement a new way to grab the RGBA32 from the GPU rendering soon