RenderHeads/UnityPlugin-AVProLiveCamera #32: Capture Results in Resolution Artifacts

Soaryn, 2021-03-24:
**Describe the bug**
When use a capture image from a Magewell Pro Capture Dual HDMI source, I noticed that the quad had a "fuzzy" appearance when looking more closely at objects or text. 

**Your Setup (please complete the following information):**
 - Unity version: 2021.2.0a9
 - AVPro Live Camera version: 2.9.1
 - Operating system version: Windows 10
 - Hardware/webcam model: Magewell Pro Capture Dual HDMI
 - Video mode (resolution, frame-rate, etc): 1920x1080; 60fps

**Screenshots**
The top illustrates a single frame in the AVLiveProCamera capture; the bottom from OBS.
[image]

This shows that the capture card is not the core problem, but my supposition is generating the backing render texture. 
_Note_: The AV illustration is the raw capture texture provided from the "select texture" in the camera component.

My settings:
[image]

Further info: the fps shows 45 as it is being shared with OBS for this test, but otherwise it is at 60fps with the same artifact problems.

kahnivore, 2021-03-24:
Hi @Soaryn 

This could possibly a small bug in the shader. 

Could you please navigate to AVProLiveCamera_YUV_HDYC2RGBA.shader and look at line 77: 

if (frac(i.uv.z) > 0.5 )

And then change the '>' symbol to a '<' symbol? 

It would be good to know if that change fixes it. 

Thanks

kahnivore, 2021-03-31:
Hi @Soaryn 

Any word on this?

closed 2021-04-22 by stale[bot]