RenderHeads/UnityPlugin-AVProLiveCamera #33: Timekeeping broken/Dropped frames not reported

BlueprintBen, 2021-06-29:
**Bug Description**
GetLastFrame/GetLastFrameUploaded doesn't increase at the rate predicted by CurrentFramerate, and GetCaptureDroppedFrames doesn't report **_any_** dropped frames during testing despite CaptureFPS dropping below CurrentFramerate. This significantly limits our ability to accurately perform timekeeping on inputs.

**My Setup:**
 - Unity version: 2020.3.12 (LTS)
 - AVPro Live Camera version: 2.9.1
 - Operating system version: Windows 10 21H1
 - Hardware/webcam model: Logitech C920 (and others)
 - Video mode (resolution, frame-rate, etc): 1920x1080@30fps (and others)

**To Reproduce**
1. Create a Unity scene with an AVProLiveCamera component configured to start using a test webcam on awake.
2. Add the attached [TimingTest.cs](https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/files/6735328/TimingTest.cs.txt) component to the same GameObject (ensure correct file extension)
3. Run the scene and wait 10 seconds for TimingTest to finish (it will stop the application itself)
4. Review the console/log to see the unaccounted-for discrepancy between time elapsed and (frames seen * framerate).

**Example Results**
[image]

BlueprintBen, 2021-09-07:
Any progress on this issue? Not being able to keep track of the 'real' index of a frame drastically reduces the utility of this plugin for us unfortunately.

kahnivore, 2021-09-10:
Hi @BlueprintBen 

Sorry for the slow response on this. 

Basically we need to investigate this further to see if this is an internal DirectShow issue or something to do with our plugin implementation, the latter which we could then look into improving. 

We will look into it ASAP just haven't had anyone available to do so. 

Thank you for your patience.

BlueprintBen, 2021-09-24:
I've tested as far back as 2.7.5 without any change, so now imagining that this is more structural. Is there a device timestamp for each frame that could at least be acquired to calculate an approximate 'realtime' frame index? Otherwise we're stuck using when we see the new frame is available, which introduces a bunch of variability/latency.

BlueprintBen, 2021-09-27:
Sorry, just a call like Plugin.GetDeviceTimestamp() isn't going to help, since it is really more data about the specific frame being accessed that I need. If the timestamp was for the active frame, at least I could calculate if it's 33ms higher than the last one (= no drops) or more (= drops), for example and make up an index from that. 

Can you explain what the GetCaptureDroppedFrames() function does report since I might still be missing some aspect of the timing.

kahnivore, 2021-10-04:
The timestamp would be for the active frame. We can add this in for you.

BlueprintBen, 2021-10-04:
OK thanks! Definitely a leg up on the timing in cases where we're trying to match things up precisely. Let know know if you want me to try out a build.

BlueprintBen, 2022-02-19:
Have come back to this and am at a bit of a loss as to why it's not feasible to implement passing along the timestamp that each DirectShow MediaSample (as long as not preview pin) has per the [DirectShow docs](https://docs.microsoft.com/en-us/windows/win32/directshow/time-stamps). Feels like a core component of DirectShow and functionality that will really allow this plugin to sync with Unity better.

That page also mentions accessing Media Times which seem even more appropriate for my purposes. Anyway would love any more insight into the blockers if possible.

Thanks!