RenderHeads/UnityPlugin-AVProLiveCamera #34: How to set Powerline Frequency (anti-flicker)?

wktfraser, 2021-07-20:
Hi Team,

Not really a bug or a feature request, just some tech support.

I'm wondering how to set PowerLine Frequency (anti-flicker) via code. It shows up in the device.ShowConfigWindow() window for Logitech C920 and BRIO webcams, but when I run through the settings using device.GetVideoSettingByIndex() it is not listed.

Hopefully I'm missing something simple :)

kahnivore, 2021-07-23:
Hi @wktfraser 

It'd be a good idea to run the CameraExplorer demo as that shows all the options we are able to detect. 

If it's not there, there's a chance we could add support for it, although, it's unlikely to be soon. 

Let us know the outcome of the above?

wktfraser, 2021-07-23:
Hi @kahnivore 

Thanks for getting back to me.

In the camera explorer, the setting does not appear under 'Settings' but if you click the 'Configure' button, this window opens where it is available.

[image]

I'm guessing that means it is not supported?

ViomerseEric, 2024-08-28:
Is this still being worked on? 

My team is getting by with a hack using an external application for now but would rather handle everything right in Unity if possible. It seems like you once had a "Freq" option exposed (saw it [here](https://discussions.unity.com/t/avpro-live-camera/477019/120) while poking around), was that just a placeholder?

Totally understand if this is not being worked on anymore though. Let me know, thanks!

Chris-RH, 2026-10-28:
Live Camera version 2.9.4 has been released.

closed 2026-11-05 by Chris-RH