RenderHeads/UnityPlugin-AVProLiveCamera #38: YUV_I420 colors displaying wrong

WCTFOX, 2021-09-24:
**Describe the bug**
YUV_I420 colors look significantly wrong when captured by Live Camera 2.6.2.  These colors were correct with Live Camera 1.6.4.  I tried swapping in the conversion shader from 1.6.4 into 2.6.2.  It gave me different colors that also weren't right.  Switching between linear and gamma color spaces also doesn't help.

**Your Setup (please complete the following information):**
 - Unity version: 2021.1.21 (and 5.5.1)
 - AVPro Live Camera version: 2.6.2
 - Operating system version: Windows 10 and Windows 11 preview
 - Hardware/webcam model: Epiphan av.io 4K
 - Video mode (resolution, frame-rate, etc): 3840 x 2160 @ 30Hz

**To Reproduce**
Steps to reproduce the behavior:
1. Hook up two PCs, one playing video into the capture card, the other receiving the feed and displaying them in a Unity program using Live Camera.
2. Show someone from the marketing department.
3. Listen to them bitch and moan because our brand colors look wrong.

**Screenshots**
I put the full 24-bit color palette through the card.  The first image is the original colors.  The second image is how they come through in OutputTexture.

[image]
[image]

Obviously going through a reduced color space you don't expect them to be identical, but it's considerably altering the hue.

WCTFOX, 2021-09-24:
It's a lot more obvious if you download the two images and switch back and forth between them. It's not that subtle a difference, but it looks much less pronounced when the images are stacked as above.

kahnivore, 2021-09-27:
Could you update to the latest version, 2.9.1, and let us know if the problem persists please?

WCTFOX, 2021-09-28:
Apologies, the version I'm using is 2.9.1.  That's the version that's showing incorrect colors.

I still had an old Readme.pdf in Assets\AVProLiveCamera; that's where I was getting the version number 2.6.2 from.

AndrewRH, 2021-10-04:
Hi,

I think this could caused by two things:

1) Gamma
2) Full/TV range

For Gamma differences, try changing the colorspace setting in Player Settings to Gamma (or Linear if it's already in Gamma).

For Full/TV range try looking at your device settings (basically there are two brightness ranges 0..255 (full) or 16..235 (tv) and any difference here can affect brightness).  You could also try toggling this here:
[image]

Lastly it could be caused by using a slightly different YCbCr matrix when converting to RGB.  

Sadly DirectShow often gives us incorrect or vague information about the matrix and range so this is a fairly common issue on some devices.  You can try upgrading the drivers of your capture device in case that helps.

You could also try disabling the option "Support Internal Format Conversion" here:
[image]

Does one of these suggestions help?

Thanks,