RenderHeads/UnityPlugin-AVProLiveCamera #39: [Bug] RAW_BGRA32 OutputTexture format is not BGRA32.

gtk2k, 2021-09-29:
I am using a Live camera (AVProLiveCamera) component.
I unchecked Auto Format in the Inspector and set the specific format to "RAW_BGRA32".
However, the format of OutputTexture is R8G8B8A8_UNorm.
I want texture format is B8G8R8A8_UNorm.

1. Why do you want B8G8R8A8? 
   Because the format required by [Unity WebRTC](https://github.com/Unity-Technologies/com.unity.webrtc) is BGRA32.   
2. What are the issues you're seeing?  --
3, What type of camera are you using? UVC camera
4. Which version of AVPro Live Camera are you using? DEMO version.

AndrewRH, 2021-10-04:
Hi @gtk2k 

You can convert to other texture formats easily in Unity.  If you create a RenderTexture in the format that you require then you can call Graphics.Blit(sourceTexture, destTexture); to convert between formats.  It's very easy.

We haven't built this functionality into AVPro Live Camera as our plugin's goal is to get video from webcam-type devices into Unity textures as quickly as possible, so we use the texture format that is closest to the webcam device's native format.

I hope that helps.

Thanks,

AndrewRH, 2021-10-04:
We will possibly rename the name of the format though so prevent confusion.