RenderHeads/UnityPlugin-AVProLiveCamera #40: UWP Support

markgrossnickle, 2021-09-30:
Unity's native camera code runs terribly on a UWP tablet. Any chance this library will gain UWP support?  If so, when?  (as we could use it on a project right now) 

Alternatively, is there a way to use AVPro Video and set the source to be the device's camera?

markgrossnickle, 2021-09-30:
(We own AVPro Video Windows)

kahnivore, 2021-10-04:
This is not something we currently support. It is likely to be added in the next major version of the plugin.

markgrossnickle, 2021-10-04:
Thanks! @kahnivore 

Do you have an ETA on the next major version?

kahnivore, 2021-10-04:
No ETA currently, sorry! 

We'd be interested to know more specifically which features you're hoping for with regards to UWP? Could you give a bit more detail about what's going wrong with the native code?

markgrossnickle, 2021-10-04:
We are working on an AR application for a UWP based tablet.  A core feature is just displaying the webcamera feed on screen.

Currently the profiler has PostLateUpdate.UpdateVideoTextures taking up an insane amount of time.  In particular YuvToRgb. 

We are just calling WebCamTexture and having it play to a texture.  RequestedFPS doesn't seem to be working (at least with this device).  

I'm thinking my next course of action will be to grab the web camera's stream via writing some native code and try to play it back in AVPro Video Windows.  Basically a poor man's version of this plugin.