RenderHeads/UnityPlugin-AVProLiveCamera #41: Hotswap doesn't work

Oneiros90, 2022-05-11:
**Describe the bug**
The hotswap feature doesn't seem to work. If I understand it correctly, this feature should allow me to disconnect my USB webcam and reconnect it back without problems.
The problem is that the webcam shows itself as active even after disconnection, while the image is frozen and does not resume after reconnection.

**Your Setup (please complete the following information):**
 - Unity version: Unity 2021.2.19f1
 - AVPro Live Camera version: 2.9.1
 - Operating system version: Windows 10
 - Hardware/webcam model: Wansview 102JD
 - Video mode (resolution, frame-rate, etc): 1080p 30fps

**To Reproduce**
Steps to reproduce the behavior:
1. Open the DefaultCameraDemo scene
2. Check the "Update Hot Swap" option in the LiveCamera component
3. Play the scene with an USB webcam on
4. Disconnect the USB webcam while playing
5. See the frozen image and the webcam in the inspector says "Displayed"
6. Reconnect the USB webcam back
7. See that the image is not resumed and the webcam shows as "Stopped"

Chris-RH, 2022-05-12:
Hi @Oneiros90,
Did you also tick the "Support Hot Swapping" option in the Manager component?

Oneiros90, 2022-05-16:
> Hi @Oneiros90, Did you also tick the "Support Hot Swapping" option in the Manager component?

It worked :D thanks

Chris-RH, 2022-05-16:
Awesome! Thanks for the reply :)

Oneiros90, 2022-06-20:
Hi @Chris-RH, sorry if I write again here, but I'm facing a weird problem with some of my clients' pc.
After I enabled both **Update Hot Swap** and **Support Hot Swap** as you suggested, some of my clients reported an heavy and noticeable performance decrease. I compiled for them a build with just the _DefaultCameraDemo_ scene with the hot swap flags disabled, then I compiled a second build with the same setup BUT with the flags enabled. They confirm that the second build is very laggy and they also recorded a video for me to see the lack of fluidity.
The weird thing is that it doesn't happen on several workstations I tried on, in my tests the performance are always good!
Any idea how to address this problem? They also sent me the Unity log but it's clean from errors...

Oneiros90, 2022-06-27:
Thank you for the suggestions @Chris-RH, I'm going with the second option :) I'll update this thread after the clients have given me feedbacks about this

Chris-RH, 2022-06-27:
Great, thank you :)

Chris-RH, 2022-07-19:
I'm going to close this for now.  Please reopen if you are still having problems with this.

closed 2022-07-19 by Chris-RH