RenderHeads/UnityPlugin-AVProLiveCamera #42: Hotswapping only updates when device is added. Not removed.

nickhudson4, 2022-06-24:
AVProLiveCameraPlugin.GetNumDevices() does not update when device is unplugged.

nickhudson4, 2022-06-27:
The device drivers are up to date. And I am using the Logitech C505 HD Webcam as well as the OBS virtual device.

Chris-RH, 2022-06-28:
I'm wondering if the OBS virtual device is confusing it perhaps. Have you tried disabling/uninstalling it?

nickhudson4, 2022-07-04:
To clarify, hotswapping it does appear to be working in the sense that I can disconnect and re-connect a device and have it start again. The issue I am having is that the devices list and NumDevices of AVProLiveCameraManager is not updating when a device is disconnected. Is it suppose to do this? Because in the manager, the update function doesn't appear to remove devices, only add. 
`	if (_supportHotSwapping && _hotSwapCounter.isOver())
			{
				if (AVProLiveCameraPlugin.UpdateDevicesConnected())
				{
					// Add any new devices
					AddNewDevices();
				}
			}
			_hotSwapCounter.incriment();`

Chris-RH, 2022-07-05:
oh right, yes, that is how it is supposed to work; its not supposed to remove them :)

closed 2022-07-12 by Chris-RH