RenderHeads/UnityPlugin-AVProLiveCamera #43: The picture of the camera is blank

Senfee-Cheng, 2022-07-19:
**Describe the bug**
The picture of the camera is blank

**Your Setup (please complete the following information):**
 - Unity version: 
 - AVPro Live Camera version: 
 - Operating system version: 
 - Hardware/webcam model: 
 - Video mode (resolution, frame-rate, etc): 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Logs**
If applicable, add error logs to help explain your problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Videos**
If applicable, add a copy of your video or the URL

Chris-RH, 2022-07-19:
Hi @Senfee-Cheng 
You've not really given enough details.

- I use the same webcam as you are using.  
- I open up a new project (I'm using Unity 2021.3.4 and DX11) and import AVPro LiveCam (version 2.9.1)
- I open up the demoscene (CameraExplorerDemo or DefaultCameraDemo are fine)
- Play Scene and select camera from dropdown menu)
- It shows the picture from my webcam

1. Does it work in the demo scenes?
2. Are your webcam drivers up  to date?
3. Does the webcam work in other software?

Chris-RH, 2022-07-26:
Closed due to lack of response, please reopen and provide more details if this is still an issue.

closed 2022-07-26 by Chris-RH