RenderHeads/UnityPlugin-AVProLiveCamera #44: Editor crash with version 2.9.2

nickhudson4, 2022-10-07:
Unity version 2021.3.10f1. Tested with fresh project on any demo scene. Also tested on empty scene only with a camera and AVProLiveCameraManager component. Editor crashes on start. Appears to happen when Plugin.Init() is called.

Stack trace: 
0x000001E8D8ADB454 (AVProLiveCamera) UnityRenderEvent
SymInit: Symbol-SearchPath: 'C:/Program Files/Unity/Hub/Editor/2021.3.10f1/Editor/Data/Mono;.;C:\Users\Nick\Documents\projects\Unity\OverMox;C:\Users\Nick\Documents\projects\Unity\OverMox\Library\BurstCache\JIT;C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor;C:\WINDOWS;C:\WINDOWS\system32;SRV*C:\websymbols*http://msdl.microsoft.com/download/symbols;', symOptions: 534, UserName: 'Nick'
OS-Version: 10.0.0
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\Unity.exe:Unity.exe (00007FF756A30000), size: 73551872 (result: 0), SymType: 'PDB', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\Unity_x64.pdb', fileVersion: 2021.3.10.32013
C:\WINDOWS\SYSTEM32\ntdll.dll:ntdll.dll (00007FFB19120000), size: 2134016 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\ntdll.dll', fileVersion: 10.0.22000.918
C:\WINDOWS\System32\KERNEL32.DLL:KERNEL32.DLL (00007FFB17910000), size: 774144 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\KERNEL32.DLL', fileVersion: 10.0.22000.708
C:\WINDOWS\System32\KERNELBASE.dll:KERNELBASE.dll (00007FFB16930000), size: 3653632 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\KERNELBASE.dll', fileVersion: 10.0.22000.918
C:\WINDOWS\System32\CRYPT32.dll:CRYPT32.dll (00007FFB16600000), size: 1449984 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\CRYPT32.dll', fileVersion: 10.0.22000.856
C:\WINDOWS\System32\ucrtbase.dll:ucrtbase.dll (00007FFB16E90000), size: 1118208 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\ucrtbase.dll', fileVersion: 10.0.22000.1
C:\WINDOWS\System32\USER32.dll:USER32.dll (00007FFB175C0000), size: 1757184 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\USER32.dll', fileVersion: 10.0.22000.832
C:\WINDOWS\System32\win32u.dll:win32u.dll (00007FFB16DA0000), size: 155648 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\win32u.dll', fileVersion: 10.0.22000.978
C:\WINDOWS\System32\GDI32.dll:GDI32.dll (00007FFB17080000), size: 167936 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\GDI32.dll', fileVersion: 10.0.22000.832
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\s3tcompress.dll:s3tcompress.dll (00007FFB026C0000), size: 180224 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\s3tcompress.dll'
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\optix.6.0.0.dll:optix.6.0.0.dll (00007FFB011A0000), size: 208896 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\optix.6.0.0.dll', fileVersion: 6.0.0.0
C:\WINDOWS\System32\gdi32full.dll:gdi32full.dll (00007FFB16770000), size: 1150976 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\gdi32full.dll', fileVersion: 10.0.22000.978
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\libfbxsdk.dll:libfbxsdk.dll (00007FFA8A480000), size: 10215424 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\libfbxsdk.dll', fileVersion: 2020.2.0.0
C:\WINDOWS\System32\msvcp_win.dll:msvcp_win.dll (00007FFB16890000), size: 643072 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\msvcp_win.dll', fileVersion: 10.0.22000.1
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\etccompress.dll:etccompress.dll (00007FFA89FA0000), size: 5066752 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\etccompress.dll'
C:\WINDOWS\System32\ADVAPI32.dll:ADVAPI32.dll (00007FFB19020000), size: 712704 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\ADVAPI32.dll', fileVersion: 10.0.22000.653
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\ispc_texcomp.dll:ispc_texcomp.dll (00007FFA9FFC0000), size: 1740800 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\ispc_texcomp.dll'
C:\WINDOWS\System32\msvcrt.dll:msvcrt.dll (00007FFB18F70000), size: 667648 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\msvcrt.dll', fileVersion: 7.0.22000.1
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\compress_bc7e.dll:compress_bc7e.dll (00007FFA9C960000), size: 1441792 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\compress_bc7e.dll'
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\FreeImage.dll:FreeImage.dll (0000000180000000), size: 6582272 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\FreeImage.dll', fileVersion: 3.18.0.0
C:\WINDOWS\System32\sechost.dll:sechost.dll (00007FFB17FF0000), size: 647168 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\sechost.dll', fileVersion: 10.0.22000.708
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\WinPixEventRuntime.dll:WinPixEventRuntime.dll (00007FFB12D00000), size: 45056 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\WinPixEventRuntime.dll', fileVersion: 1.0.1812.6001
C:\WINDOWS\System32\RPCRT4.dll:RPCRT4.dll (00007FFB17ED0000), size: 1179648 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\RPCRT4.dll', fileVersion: 10.0.22000.675
C:\WINDOWS\System32\WS2_32.dll:WS2_32.dll (00007FFB16FB0000), size: 454656 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\WS2_32.dll', fileVersion: 10.0.22000.1
C:\WINDOWS\System32\SHELL32.dll:SHELL32.dll (00007FFB18090000), size: 8093696 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\SHELL32.dll', fileVersion: 10.0.22000.918
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\umbraoptimizer64.dll:umbraoptimizer64.dll (00007FFA9BF90000), size: 1187840 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\umbraoptimizer64.dll'
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\OpenImageDenoise.dll:OpenImageDenoise.dll (00007FFA828A0000), size: 48848896 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\OpenImageDenoise.dll', fileVersion: 1.4.2.0
C:\WINDOWS\System32\SHLWAPI.dll:SHLWAPI.dll (00007FFB17020000), size: 380928 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\SHLWAPI.dll', fileVersion: 10.0.22000.1
C:\WINDOWS\System32\ole32.dll:ole32.dll (00007FFB171B0000), size: 1679360 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\ole32.dll', fileVersion: 10.0.22000.120
C:\WINDOWS\System32\combase.dll:combase.dll (00007FFB17B30000), size: 3641344 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\combase.dll', fileVersion: 10.0.22000.918
C:\WINDOWS\SYSTEM32\MSVCP140.dll:MSVCP140.dll (00007FFB071F0000), size: 581632 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\MSVCP140.dll', fileVersion: 14.32.31326.0
C:\WINDOWS\SYSTEM32\VCRUNTIME140.dll:VCRUNTIME140.dll (00007FFB07460000), size: 110592 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\VCRUNTIME140.dll', fileVersion: 14.32.31326.0
C:\WINDOWS\System32\IMM32.dll:IMM32.dll (00007FFB18900000), size: 200704 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\IMM32.dll', fileVersion: 10.0.22000.1
C:\WINDOWS\System32\SETUPAPI.dll:SETUPAPI.dll (00007FFB18B00000), size: 4636672 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\SETUPAPI.dll', fileVersion: 10.0.22000.469
C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\tbb12.dll:tbb12.dll (00007FFAD6610000), size: 434176 (result: 0), SymType: '-exported-', PDB: 'C:\Program Files\Unity\Hub\Editor\2021.3.10f1\Editor\tbb12.dll'
C:\WINDOWS\System32\OLEAUT32.dll:OLEAUT32.dll (00007FFB174E0000), size: 876544 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\OLEAUT32.dll', fileVersion: 10.0.22000.1
C:\WINDOWS\SYSTEM32\OPENGL32.dll:OPENGL32.dll (00007FFABB480000), size: 1052672 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\OPENGL32.dll', fileVersion: 10.0.22000.708
C:\WINDOWS\System32\WINTRUST.dll:WINTRUST.dll (00007FFB16D30000), size: 425984 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\System32\WINTRUST.dll', fileVersion: 10.0.22000.918
[... cut ...]
0x00007FF7576119B6 (Unity) ExecutePlayerLoop
0x00007FF7576177A9 (Unity) PlayerLoop
0x00007FF758562E69 (Unity) PlayerLoopController::UpdateScene
0x00007FF75854E230 (Unity) PlayerLoopController::EnterPlayMode
0x00007FF75855DF67 (Unity) PlayerLoopController::SetIsPlaying
0x00007FF758560412 (Unity) Application::TickTimer
0x00007FF7589AFF5A (Unity) MainMessageLoop
0x00007FF7589B480B (Unity) WinMain
0x00007FF759D420CE (Unity) __scrt_common_main_seh
0x00007FFB179254E0 (KERNEL32) BaseThreadInitThunk
0x00007FFB1912485B (ntdll) RtlUserThreadStart

Chris-RH, 2022-10-10:
Hi @nickhudson4,

I have just tested this on my PC using Unity 2021.3.10 and AVPro Live Camera 2.9.2 and I've not had an issue with it.  

Have you tried closing Unity, deleting the AVPro files, starting a new project in Unity, re-importing the AVPro files and then running the Camera Explorer Demo?

nickhudson4, 2022-10-10:
Hey!

This was tested on my main project as well as a fresh projects. Was also testing on my clients pc with the same result.

Were there any plugin changes with the version? Or were all the changes Unity side?

Chris-RH, 2022-10-10:
Did you re-download the files in your package manager?

nickhudson4, 2022-10-10:
Yes

Chris-RH, 2022-10-10:
Are you using D3D11 or 12?
Which version of Windows is it?
Which camera are you trying to connect to?
Does this occur when you try all of the demo scenes as soon as you run them?

nickhudson4, 2022-10-10:
- Direct3D11
- Windows 11 21H2
- Never get the chance to connect to a camera. The crash occurs at the plugin init from the AVProLiveCameraManager.
- Yes. Also occurs in empty scene with only the manager component added.

Chris-RH, 2022-10-11:
Thank you.  
Was the other machine that you tested it on also Windows 11?
Was there a previous version of the asset that did work for you?

Chris-RH, 2022-10-12:
Does this still occur if no camera is connected to your PC?

nickhudson4, 2022-10-12:
Figured out that it is caused by the NDI tools software which creates and NDI virtual cam  https://www.ndi.tv/tools/

Chris-RH, 2022-10-12:
OK, that's good to hear. So have you got it working again?

nickhudson4, 2022-10-13:
Most of our user base will be pairing NDI tools with AVProLive since they both reside in the streaming space. So it will be a pretty big issue for us. Is it something that can be checked out?

Chris-RH, 2022-10-14:
Yes, now that the cause of the problem has been found, it shouldn't be too difficult to fix.

Chris-RH, 2022-10-14:
The next version has been released, please let us know if this has not fixed the issue for you.

nickhudson4, 2022-10-14:
Tested and working. Thanks!

closed 2022-10-17 by Chris-RH