RenderHeads/UnityPlugin-AVProLiveCamera #46: Black screen with pink line that fades into green.

Seratonin, 2023-01-17:
Please DO NOT LINK / ATTACH YOUR PROJECT FILES HERE

**Describe the bug**
Black screen that fades into pink going down vertically. It would disappear for a couple milliseconds and re-appear. With that being said, there's an upside down message that reappears. It says AVPRO Trial. I had to zoom in to take a picture.

**Your Setup (please complete the following information):**
 - Unity version: 2021.3.16f1
 - AVPro Live Camera version: 2.9.3
 - Operating system version: Windows 11
 - Hardware/webcam model: Elgato Capture HD60 S+
 - Video mode (resolution, frame-rate, etc): 1080p, 60fps, Color mode is Bypass, 

**To Reproduce**
Steps to reproduce the behavior:
1. Opened DefaultCameraDemo
2. Pressed Play button
3. Selected Game Capture HD60 S+, 1920x1080, 60fps, YUV_YUY2 (Issue occurs on all formats)
4. Results with black screen that fades into pink going down vertically. It would disappear for a couple milliseconds and re-appear. With that being said, there's an upside down message that reappears. It says AVPRO Trial. I had to zoom in to take a picture.

[image]
[image]

Chris-RH, 2023-01-17:
The scrolling line looks like its to do with the watermark.  

Do you get any image from your camera at all?
Have you tried the other formats?
Are you using D3D11 or D3D12?

Seratonin, 2023-01-21:
Yes I get a clear image but with a weird line going across.
The issue occurs on all formats.
The thing is, by default Auto Graphics API for Windows is enabled. When I disable it, the only option listed by default is Direct3D11. So I assume Direct3D11 is the answer to your question.

Chris-RH, 2023-02-06:
As mentioned, the line going across will be the watermark.  If you are still not sure, please can you post a clearer image with it against a background.