RenderHeads/UnityPlugin-AVProLiveCamera #47: Frame rate not consistent

georgeUAMTEC, 2023-02-22:
Hi Guys

**Describe the bug**
I have been having frame stutter and very bumpy frame rates. Using a 4k camera which we provide to our clients with our built app
using your API.

Even at 1080 it seems a little better however after a little time of running it starts to do the same thing. I am running this on an Alien ware RTX, Where our clients have slightly lower spec machines.

Our cam is custom provided from factory.

We test this using Win 10 Camera app running at 4k and recording at Highest possible and does not give us this issue.
We also use you capture app and that tends to drop it more. We did our best to optimise where we can.

Are there any recommended settings or ways to optimise when they are running at 4k?

**Your Setup (please complete the following information):**
 - Unity version: 2020.3.42f1
 - AVPro Live Camera version: 2.9.3 (Current)
 - Operating system version: Windows 10
 - Hardware/webcam model: Ailipu 4K USB Camera - Camera, Audio
 - Video mode (resolution, frame-rate, etc): 
 3840 2160 RAW_BGRA32 Framerates 25/29/23 What I get! I attached a screenshot

**Logs**
I don't get error or logs. Not that I can see

**Screenshots**
[image]

It seems to work really heavy when I am running it in Unity. There is no issue when running the Win App.
[image]

**Videos**
I had to lower there resolution to reduce file size. They have been both captured at 4k

2 videos

Unity at 4k stream/recording
https://user-images.githubusercontent.com/126118489/220782810-7383b53c-8817-45e7-b453-31637e1d6c58.mp4

Win Camera app 4k Stream/recording
https://user-images.githubusercontent.com/126118489/220782819-29d344b6-67ed-4b76-ba2e-43afdfe2e07b.mp4

Chris-RH, 2023-02-23:
Out of interest, what FPS do you get when you build the project on your PC, rather than running it in Unity Editor?  Unity will add to the resource overheads, so is not a true test.

Other things to help improve FPS:

- You should check the settings of the camera to make sure it isn’t doing unnecessary processing which reduces the frame rate. Disable options like “low light compensation”, disable auto exposure and reduce exposure settings. You can access camera settings via the CameraExplorer demo (click “Configure”), or usually via the camera setup application on the Start Menu.

- Another possibility is that the data bus is overloaded.  You may need to plug devices into different ports so they are on a different USB bus. 

- You could also check that the checkboxes for Hot Swapping support and Settings Update are not checked in the AVProLiveCameraManager and AVProLiveCamera components. On some device drivers polling for device connection status and checking the settings can really slow things down and could be done manually instead.

georgeUAMTEC, 2023-02-28:
Hi Chris,

Thankyou for your response!

sorry I couldn't back to you earlier. I would look into it today and let you know how that goes.

Hopefully that will improve this. And will update you on it:)