RenderHeads/UnityPlugin-AVProLiveCamera #49: Problems with Elgato Facecam Pro

ToxMox, 2023-05-09:
Chris over on the Unity forum asked me to post this here as it will likely require you guys acquiring an Elgato Facecam Pro (which was finally back in stock on Elgato's website the other day but now out of stock again)

**Describe the bug**
Having some problems with the Elgato Facecam Pro:

    The 3840x2160 [MJPG] mode is stuck at 30hz even when I specify 60hz
    The 3840x2160 [Unknown] mode just doesn't work. I believe these "unknown" modes are h264 based on the attached DirectShow Capture Capabilities output. The 1920x1080 [Unknown] mode works

**Screenshot showing [Unknown] mode not working:**
[image]

**Screenshot showing only getting 30hz:**
[image]

**Directshow Capabilities:**
[ElgatoFacecamPro.txt](https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/files/11433659/ElgatoFacecamPro.txt)

**Relevant forum post:**
[image]

**My Setup:**
 - Unity version: 2021.3.10f1
 - AVPro Live Camera version: 2.9.3
 - Operating system version: Windows 10
 - Hardware/webcam model: Elgato Facecam Pro

Chris-RH, 2023-05-10:
Thanks, this will be helpful in case anyone else runs into the same problem as it gives a few things to try.
We hopefully have one ordered and on its way to the office, but I'm not sure exactly when it will arrive.

RichRH, 2023-08-10:
Hi @ToxMox,

Thanks for reporting this - we finally managed to get our hands on the ellusive Facecam Pro and have reproduced the issues you're seeing.

Unfortunately there's no quick fix - 4K MJPG does indeed seem to be limited to 30hz (I suspect due to API/bandwidth limitations as 60hz is definitely the value being passed to the device - confirmed in debugger) and as you correctly suggest, the 'Unknown' format is indeed H.264, which the plugin doesn't directly support.

I think we're straddling 'enhancement' rather than 'bug' at this point but am hoping some work I'm doing on the other plugins _may_ pave the way for the H.264 route to work, although fundamentally we really need to update the base API used by Live Camera to fully address this - on the TODO list!

Cheers,

ToxMox, 2023-08-10:
Excellent, thank you! Very much looking forward to the fixes 😄

ToxMox, 2023-11-27:
Hi there. I'm curious if there has been any progress or update on getting Media Foundation cameras supported?

ToxMox, 2024-02-29:
@RichRH any update on this?

shealan, 2024-03-17:
Any updates on this? We have just invested heavily in these cameras and discovering they don't play nicely with this package is going to cause a lot of problems for us.

RichRH, 2024-03-18:
Hi @ToxMox/@shealan,

I'm afraid there's no update on this as of yet - 4K/30 is currently the limitation. We got as far as acquiring the camera and confirming the limitation but have been slammed with other work. It's still on the agenda but unfortunately couldn't give a timeframe at this point as it's pretty much a rewrite of the plugin (for Media Foundation support).

Cheers,

HectorZap, 2024-07-07:
Hello, I have the same issue. Any updates/roadmap on this please?
It's very frustrating having bought both the camera and the plugin to get the maximum possible quality and see it capped to 30 fps.
Any update, and hopefully a timeline, would be very welcome. Thanks,

RichRH, 2024-07-12:
Hi @HectorZap,

We share your frustration but as a small team have to choose our battles carefully, having to prioritise work based on severity and demand. Demand for this issue is certainly increasing and I can tell you that fundamental work has started on rewriting this plugin, but it will take some time and is contigent on a clear runway (contract projects etc..). I can't make any promises but late Autumn/early Winter is a realistic target.

Cheers,

HectorZap, 2024-07-22:
Thanks a lot for your answer, and for your hard work! Looking forward to the update,

HectorZap, 2024-11-15:
Hello! Have you made any progress on this? Looking forward to get 4k 60 fps from our ElGato Facecam Pro

RichRH, 2024-11-18:
Hi @HectorZap,

We're actually in the process of rewriting all (Windows) plugins to better homogenise them, with Video being priority, then Live Camera (as we actually want to use the Elagato Facecam in some of our own work @ 4K/60) - progress is good but we're not quite there yet. We're spinning a lot of plates so it's hard to give estimates but we'd hope to be internally testing still within the Winter months.

Cheers,

shealan, 2025-01-14:
Hey @RichRH are you able to share any progress on this?

RichRH, 2025-01-22:
Hi @shealan,

Thanks for your message - we are progressing but the Video rewrite is taking a bit longer than planned (we're a small team wearing many hats), although a lot of those improvements with this new setup will naturally flow into Live Camera, which is still our next priority (for our internal use too). With a bit of luck we're still talking Winter months but it may slip a bit into Spring depending on workload etc..

Cheers,

ToxMox, 2025-04-26:
Hello stale bot. I'd like to keep this issue open, thank you

shealan, 2025-08-24:
Also keen for an update

ToxMox, 2025-09-05:
> As ever, this is still a work in progress.

I first submitted this issue in March of 2023 on the unity forum. These delays have been pretty disappointing. Your reply here is a bit concerning.

Soaryn, 2026-08-20:
I can only imagine how much work is required for the rewrite, but the last mention of the win dev on annual leave for nearly a year now with no additional info is quite ironic.

I assume there is no additional info?

RichRH, 2026-10-28:
Hi all,

Firstly, an apology for being very quiet here - it's hard to emphasise how small a team we have vs how many plates we have spinning, especially now, given the massively evolving software landscape!

I made the mistake of suggesting potential release dates, which I appreciate people latch onto, and again apologise, as at the time those dates felt realistic but then other priorities came along and they fell by the wayside.

I'm not going to jinx things/make the mistake of suggesting any more dates but as an update; the AVPro Video rewrite is progressing well, I have 4k60 working with the Elgato Facecam Pro internally, and a new version of AVPro Live Camera is taking shape and still next on the list for release _after_ the new version of Video.

I appreciate there will be frustration, but can only apologise again and reiterate these rewrites are also being developed for our own internal needs, so do share in that frustration.

As a heads up, there should be an update going out to Live Camera soon to fix a few long-standing bugs, **but is not the rewritten version and so camera support remains as was**.

Cheers,