RenderHeads/UnityPlugin-AVProLiveCamera #50: PolyPop camera crashes Unity when exiting playmode

Gabana, 2023-05-10:
Hi,

I found a bug that is making Unity crash. Whenver i select the Polypop camera (it got installed when i installing the software https://www.polypoplive.com/) and i exit Playmode, Unity crashes. I can enter playmode and switch between my webcams or virtual cameras without it crashing but when i stop/exit Playmode, it crashes all the times.

Thanks!

**Setup:**
 - Unity version:  2022.2.0b11 and  2022.2.1f1.
 - AVPro Live Camera version: 2.9.3
 - Operating system version: Windows 11
 - Hardware/webcam model: 
 - Video mode (resolution, frame-rate, etc): 

**To Reproduce**
1. Install PolyPop software (https://www.polypoplive.com/)
2. Launch Camera Explorer Demo scene
3. Start the PolyPop Camera
4. Stop Unity (i.e. exit Playmode)
5. Observe Unity crashes

RichRH, 2023-08-11:
Hi @Gabana,

Thanks for reporting this - I reproduced the issue using the steps you provided and found the problem. Fix will be in the next version.

Cheers,

Chris-RH, 2026-10-28:
Live Camera version 2.9.4 has been released.

closed 2026-11-05 by Chris-RH