RenderHeads/UnityPlugin-AVProLiveCamera #53: The client consistently crashes on this PC device

jinchao90, 2024-04-24:
The client consistently crashes on this PC device. However, after removing the AV plugin and related modules, it starts up normally. Please address this issue.
```
AS: AutoStreaming module initializing.
Initialize engine version: 2021.3.1f1c1 (0cb3bd085a01)
[Subsystems] Discovering subsystems at path E:/soft/324b12b3756b5518GpxYnMabQjPwmAB/1221XR_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1; jobified=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: Intel(R) UHD Graphics 730 (ID=0x4692)
    Vendor:   Intel
    VRAM:     7979 MB
    Driver:   31.0.101.5186
FMOD failed to initialize any audio devices, running on emulated software output with no sound. Please check your audio drivers and/or hardware for malfunction.
Begin MonoManager ReloadAssembly
- Completed reload, in  0.095 seconds
<RI> Initializing input.
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.
<RI> Initialized touch support.
UnloadTime: 0.827700 ms
[AVProLiveCamera] version 2.9.1 initialised. Intel(R) UHD Graphics 730
```

jinchao90, 2024-04-24:
There is no such issue on other window devices. Do you have any hardware requirements for PC devices, or do you need other software environments

jinchao90, 2024-04-24:
Looking forward to your reply ：）

jinchao90, 2024-04-25:
Thank you for your interest ：）
Here are the details

Setup:
- Unity version: 2021.3.1f1c1
- AVPro Live Camera version: 2.9.1
- Operating system version: Windows 11
- Hardware/webcam model:
- Video mode (resolution, frame-rate, etc):
To Reproduce

1. open project（exe）
2. project crash

Here is all the log information
```
Mono path[0] = 'C:/Users/Admin/Downloads/dev-1.2.0-202404251135/DPXR_Data/Managed'
Mono config path = 'C:/Users/Admin/Downloads/dev-1.2.0-202404251135/MonoBleedingEdge/etc'
Found 1 interfaces on host : 0) 192.168.50.167
Multi-casting "[IP] 192.168.50.167 [Port] 55469 [Flags] 2 [Guid] 2572512498 [EditorId] 2458535500 [Version] 1048832 [Id] WindowsPlayer(2,DESKTOP-60JW8RO) [Debug] 0 [PackageName] WindowsPlayer [ProjectName] DPXR" to [225.0.0.222:54997]...
AS: AutoStreaming module initializing.
Initialize engine version: 2021.3.1f1c1 (0cb3bd085a01)
[Subsystems] Discovering subsystems at path C:/Users/Admin/Downloads/dev-1.2.0-202404251135/DPXR_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1; jobified=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: Intel(R) UHD Graphics 730 (ID=0x4692)
    Vendor:   Intel
    VRAM:     7979 MB
    Driver:   31.0.101.5186
FMOD failed to initialize any audio devices, running on emulated software output with no sound. Please check your audio drivers and/or hardware for malfunction.
Begin MonoManager ReloadAssembly
- Completed reload, in  0.063 seconds
<RI> Initializing input.
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.
<RI> Initialized touch support.
UnloadTime: 2.303200 ms

[AVProLiveCamera] version 2.9.1 initialised. Intel(R) UHD Graphics 730
UnityEngine.StackTraceUtility:ExtractStackTrace ()
UnityEngine.DebugLogHandler:LogFormat (UnityEngine.LogType,UnityEngine.Object,string,object[])
UnityEngine.Logger:Log (UnityEngine.LogType,object)
UnityEngine.Debug:Log (object)
RenderHeads.Media.AVProLiveCamera.AVProLiveCameraManager:Init () (at D:/XR/client-pc/XRClient/Assets/AVProLiveCamera/Scripts/Components/AVProLiveCameraManager.cs:89)
RenderHeads.Media.AVProLiveCamera.AVProLiveCameraManager:Start () (at D:/XR/client-pc/XRClient/Assets/AVProLiveCamera/Scripts/Components/AVProLiveCameraManager.cs:74)

(Filename: D:/XR/client-pc/XRClient/Assets/AVProLiveCamera/Scripts/Components/AVProLiveCameraManager.cs Line: 89)
```

jinchao90, 2024-04-25:
I noticed that the crash log, and the initial judgment is that the initialization failed to get the camera audio, causing the project to crash, and I didn't see any other additional log information
[image]

jinchao90, 2024-04-25:
Looking forward to your reply and help ：）

jinchao90, 2024-04-28:
In the past two days, I have done a lot of tests, reinstalled the system and drivers, and found that it works normally, but after I installed NDI, I found that the previous problem appeared again, but I followed the above steps to retest it to determine if it was caused by NDI, and found that everything is normal, so NDI is not the root cause of this problem. I'm trying to minimize the external influence, the current situation is the same computer, the same project, I am running fine in the unity editor, but the packaged PC does not run, this problem occurs every time

jinchao90, 2024-04-28:
Using the same computer as described above, the following is the latest log

```
Mono path[0] = 'D:/PAC/xr291-prue-load/release-1.2.1-202404281218/DPXR_Data/Managed'
Mono config path = 'D:/PAC/xr291-prue-load/release-1.2.1-202404281218/MonoBleedingEdge/etc'
AS: AutoStreaming module initializing.
Initialize engine version: 2021.3.1f1c1 (0cb3bd085a01)
[Subsystems] Discovering subsystems at path D:/PAC/xr291-prue-load/release-1.2.1-202404281218/DPXR_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1; jobified=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: Intel(R) UHD Graphics 730 (ID=0x4692)
    Vendor:   Intel
    VRAM:     7979 MB
    Driver:   31.0.101.5186
FMOD failed to initialize any audio devices, running on emulated software output with no sound. Please check your audio drivers and/or hardware for malfunction.
Begin MonoManager ReloadAssembly
- Completed reload, in  0.065 seconds
<RI> Initializing input.
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.
<RI> Initialized touch support.
UnloadTime: 1.153200 ms
[AVProLiveCamera] version 2.9.1 initialised. Intel(R) UHD Graphics 730
```

jinchao90, 2024-04-28:
The code crashes here and the exe crashes at this point, but it doesn't appear in the Unity editor

[image]

Chris-RH, 2024-05-09:
Can you try upgrading to the latest version of AVPro Live Camera and retesting please :)

closed 2024-12-05 by Chris-RH