RenderHeads/UnityPlugin-AVProLiveCamera #55: Camera feed freezes sometimes

gekkyo, 2024-12-05:
### AVPro Live Camera edition

Full

### AVPro Live Camera Version

2.9.3

### Your question

Hi. I have purchased the plugin and am using it for a commercial signage project.
It is working great, but I'm encountering a issue and would appreciate some help.

Every 2 to 3 days, the camera feed freezes, and I've been receiving inquiries from my client.
This signage operates for about 8 hours a day, and restarting the machine or the application during operation is not allowed,
which is causing some difficulties.

I am working on detecting when the freeze occurs in order to automatically reset the camera feed.
In that case, what would be the best approach to detect the freeze?

In the similar [issue](https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/issues/25) Andrew suggested to monitor the texture update index. 
Specifically, how do I do that?

Chris-RH, 2024-12-05:
We need some more information please

1. Unity version:
2. Windows version:
3. Graphics API:
4. Hardware and/or Webcam name/model number:
5. Computer hardware that app is running on, to include CPU, GPU, RAM etc):
6. Details of video mode you are using, to include: resolution, frame rate, and colour mode:
7. Does the freezing occur within the 8 hour window of operation or is it only if running for longer than 8 hours?
8. Have they tried starting restarting it every morning?

RichRH, 2024-12-06:
Hi @gekkyo,

I believe Andrew was referring to monitoring something like 'AVProLiveCameraPlugin.GetLastFrameUploaded(deviceIndex)' or 'device.DisplayFPS' which will only be changing if new frames are being fed through.

Therefore you can run some kind of timer and if there hasn't been a change within a certain time period (depends how fine you want to be but e.g. a second) then you can detect a freeze.

Cheers,

gekkyo, 2024-12-18:
@RichRH 
I have confirmed that monitoring "AVProLiveCameraPlugin.GetLastFrameUploaded(deviceIndex)" allows me to detect a freeze. So, I will incorporating the feed reset process when that happens.
Thanks for your support!

closed 2024-12-18 by gekkyo