RenderHeads/UnityPlugin-AVProLiveCamera #56: [Bug] Not Support DX12

SinForLove, 2025-01-09:
### Unity version

6000.0.25

### AVPro Live Camera edition

Full

### AVPro Live Camera Version

2.9.3

### Which Windows version(s) are you using?

Windows 10

### Which graphics API(s) are you using?

Direct3D 12

### Hardware/Webcam model

It work when dx11 . But if I switch to DX12 ,it not work but still can get device info.

### Video mode

Any mode

### The issue

Use unity 6 hdrp and dx12 api;

Chris-RH, 2025-01-09:
Hi, 
[AVPro Live Camera does not yet support DX12](https://github.com/RenderHeads/UnityPlugin-AVProLiveCamera/issues/31).

closed 2025-01-09 by Chris-RH

houseofsecrets, 2025-03-26:
Is a rewrite really necessary for you to get this working. It's kind of annoying to pay this much for your plugins and then getting hanged out to dry when it comes to fixing bugs because you want to rewrite AVPro.

Guess it's time to look for another alternative to WebcamTexture, again....Shame